NordicSemiconductor/nRF-IEEE-802.15.4-radio-driver #129: Simple example TX/RX

ochipara, 2018-06-13:
Hi,

I am trying to build a simple example for using the stack to send and receive. I am trying to use only RAAL_SINGLE_PHY. However, I am running into some problems since the documentation is a bit outdated. For example -- no information is provided about nrf_802154_rsch. If any of you could point me in the right direction, I would appreciate it.

Thanks!

hubertmis, 2018-06-14:
Thanks for reporting this issue. We are in the middle of adding new feature to the master branch and haven't updated wiki yet. But we should have.
If you don't need the 'delayed transmission' feature, try to use 4fbe9d9d29291921295c1df56b6ffb6f44d53af7 commit. Wiki should be up-to-date with this one. We'll update the docs soon.

hubertmis, 2018-10-11:
I've updated the wiki. It should contain correct file list.

closed 2018-10-11 by hubertmis