NordicSemiconductor/nRF-IEEE-802.15.4-radio-driver #162: Driver crashes

Nopik, 2018-10-12:
For some time already I'm hunting down some crashes in my nRF52840 application using this driver. I've finally managed to reduce the problem down to pretty simple application - I made it available at: https://github.com/Nopik/nrf_radio_crash

It generally requires 2 PCA10056 boards, turns one into sender of simple packets (about 50 per second), the other node is receiver. Typically receiver crashes a lot, when receiving the packets. I added much more elaborated info on the README.md file on the repository.

Typically crashes occur within first few seconds - and if receiver did not crashed after running for 10 seconds or more, it will usually run for long time, like 20-40 minutes or even few hours. Therefore, receiver part self-resets after receiving 500 packets (roughly 10 seconds). On my PCA10056 boards, they don't survive more than few cycles (<1 minute).

Please download the app and try to reproduce the problem. There is a chance, that I'm using the driver in some incorrect way, though the reception code is pretty simple.

hubertmis, 2018-10-15:
Thanks for reporting the problem.
Could you provide more information? 
What do you mean 'crashes'? Is it assertion fault? CPU exception? Application error? Infinite loop? Or anything else?
Could you provide backtrace captured when your program crashes?
Also, it would be helpful if you could report the driver debug log: https://github.com/NordicSemiconductor/nRF-IEEE-802.15.4-radio-driver/wiki/Debug-mode

Nopik, 2018-10-15:
Well, generally, when the problem happens, all FreeRTOS tasks are stopping execution. When I added led blinking in radio IRQ driver, it also stopped to blink. From my perspective, CPU is just halted - or, at least, if there is any activity there, I cannot tell. Generally it looks very much like memory corruption.

If you run the application provided, there is a FreeRTOS idle hook, which is called very often, and toggles led (https://github.com/Nopik/nrf_radio_crash/blob/master/src/crasher/application_crasher.cpp#L110). When problem occurs, led stops to blink, too.

While I do have a number of FreeRTOS hooks installed, none of them seem to fire (or at least, they don't have time to print the logs).

I will try to provide the log. Last time I was playing with it, however, the problems appeared outside of the IRQ handler.

Nopik, 2018-10-15:
So, here are some debugs, pointing to hardfault:

```
arm-none-eabi-gdb build/nrf52840_xxaa.out
GNU gdb (GNU Tools for ARM Embedded Processors) 7.8.0.20150604-cvs
Copyright (C) 2014 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "--host=x86_64-apple-darwin10 --target=arm-none-eabi".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
No symbol table is loaded.  Use the "file" command.
Reading symbols from build/nrf52840_xxaa.out...done.
(gdb) target remote localhost:2331
Remote debugging using localhost:2331
HardFault_Handler ()
    at ../../../../third_party/nRF5_SDK_for_Thread_and_Zigbee/modules/nrfx/mdk/gcc_startup_nrf52840.S:300
300	    b       .
(gdb) bt
#0  HardFault_Handler ()
    at ../../../../third_party/nRF5_SDK_for_Thread_and_Zigbee/modules/nrfx/mdk/gcc_startup_nrf52840.S:300
#1  <signal handler called>
#2  0x00001a02 in PendSV_Handler ()
    at ../../../../third_party/nRF5_SDK_for_Thread_and_Zigbee/external/freertos/portable/GCC/nrf52/port.c:104
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
(gdb) set print elements 0
(gdb) set print repeats unlimited
(gdb) set pagination off
(gdb) set height unlimited
(gdb)
(gdb) p/z nrf_802154_debug_log_buffer
$1 = {0x01000001, 0x02040001, 0x02040002, 0x01020001, 0x01020002, 0x02050001, 0x02050002, 0x01000002, 0x01000001, 0x02040001, 0x02040002, 0x01020001, 0x01020002, 0x02050001, 0x02050002, 0x01000002, 0x01000001, 0x02040001, 0x02040002, 0x010a0001, 0x00050001, 0x02040001, 0x02040002, 0x02050001, 0x02050002, 0x00050002, 0x010a0002, 0x02050001, 0x02050002, 0x01000002, 0x01000001, 0x02040001, 0x02040002, 0x01020001, 0x01020002, 0x02050001, 0x02050002, 0x01000002, 0x01000001, 0x02040001, 0x02040002, 0x01020001, 0x01020002, 0x02050001, 0x02050002, 0x01000002, 0x01000001, 0x02040001, 0x02040002, 0x010a0001, 0x00050001, 0x02040001, 0x02040002, 0x02050001, 0x02050002, 0x00050002, 0x010a0002, 0x02050001, 0x02050002, 0x01000002, 0x01000001, 0x02040001, 0x02040002, 0x01020001, 0x01020002, 0x02050001, 0x02050002, 0x01000002, 0x01000001, 0x02040001, 0x02040002, 0x01020001, 0x01020002, 0x02050001, 0x02050002, 0x01000002, 0x01000001, 0x02040001, 0x02040002, 0x010a0001, 0x00050001, 0x02040001, 0x02040002, 0x02050001, 0x02050002, 0x00050002, 0x010a0002, 0x02050001, 0x02050002, 0x01000002, 0x01000001, 0x02040001, 0x02040002, 0x01020001, 0x01020002, 0x02050001, 0x02050002, 0x01000002, 0x01000001, 0x02040001, 0x02040002, 0x01020001, 0x01020002, 0x02050001, 0x02050002, 0x01000002, 0x01000001, 0x02040001, 0x02040002, 0x010a0001, 0x00050001, 0x02040001, 0x02040002, 0x02050001, 0x02050002, 0x00050002, 0x010a0002, 0x02050001, 0x02050002, 0x01000002, 0x01000001, 0x02040001, 0x02040002, 0x01020001, 0x01020002, 0x02050001, 0x02050002, 0x01000002, 0x01000001, 0x02040001, 0x02040002, 0x01020001, 0x01020002, 0x02050001, 0x02050002, 0x01000002, 0x01000001, 0x02040001, 0x02040002, 0x010a0001, 0x00050001, 0x02040001, 0x02040002, 0x02050001, 0x02050002, 0x00050002, 0x010a0002, 0x02050001, 0x02050002, 0x01000002, 0x01000001, 0x02040001, 0x02040002, 0x01020001, 0x01020002, 0x02050001, 0x02050002, 0x01000002, 0x01000001, 0x02040001, 0x02040002, 0x01020001, 0x01020002, 0x02050001, 0x02050002, 0x01000002, 0x01000001, 0x02040001, 0x02040002, 0x010a0001, 0x00050001, 0x02040001, 0x02040002, 0x02050001, 0x02050002, 0x00050002, 0x010a0002, 0x02050001, 0x02050002, 0x01000002, 0x01000001, 0x02040001, 0x02040002, 0x01020001, 0x01020002, 0x02050001, 0x02050002, 0x01000002, 0x01000001, 0x02040001, 0x02040002, 0x01020001, 0x01020002, 0x02050001, 0x02050002, 0x01000002, 0x01000001, 0x02040001, 0x02040002, 0x010a0001, 0x00050001, 0x02040001, 0x02040002, 0x02050001, 0x02050002, 0x00050002, 0x010a0002, 0x02050001, 0x02050002, 0x01000002, 0x01000001, 0x02040001, 0x02040002, 0x01020001, 0x01020002, 0x02050001, 0x02050002, 0x01000002, 0x01000001, 0x02040001, 0x02040002, 0x01020001, 0x01020002, 0x02050001, 0x02050002, 0x01000002, 0x01000001, 0x02040001, 0x02040002, 0x010a0001, 0x00050001, 0x02040001, 0x02040002, 0x02050001, 0x02050002, 0x00050002, 0x010a0002, 0x02050001, 0x02050002, 0x01000002, 0x01000001, 0x02040001, 0x02040002, 0x01020001, 0x01020002, 0x02050001, 0x02050002, 0x01000002, 0x01000001, 0x02040001, 0x02040002, 0x01020001, 0x01020002, 0x02050001, 0x02050002, 0x01000002, 0x01000001, 0x02040001, 0x02040002, 0x010a0001, 0x00050001, 0x02040001, 0x02040002, 0x02050001, 0x02050002, 0x00050002, 0x010a0002, 0x02050001, 0x02050002, 0x01000002, 0x01000001, 0x02040001, 0x02040002, 0x01020001, 0x01020002, 0x02050001, 0x02050002, 0x01000002, 0x01000001, 0x02040001, 0x02040002, 0x01020001, 0x01020002, 0x02050001, 0x02050002, 0x01000002, 0x01000001, 0x02040001, 0x02040002, 0x010a0001, 0x00050001, 0x02040001, 0x02040002, 0x02050001, 0x02050002, 0x00050002, 0x010a0002, 0x02050001, 0x02050002, 0x01000002, 0x01000001, 0x02040001, 0x02040002, 0x01020001, 0x01020002, 0x02050001, 0x02050002, 0x01000002, 0x01000001, 0x02040001, 0x02040002, 0x010 [... cut ...]
ERROR: Can not read register 8 (R8) while CPU is running
ERROR: Can not read register 9 (R9) while CPU is running
ERROR: Can not read register 10 (R10) while CPU is running
...
ERROR: Can not read register 62 (FPS29) while CPU is running
ERROR: Can not read register 63 (FPS30) while CPU is running
ERROR: Can not read register 64 (FPS31) while CPU is running
Read 4 bytes @ address 0x00000000 (Data = 0x20040000)
Read 4 bytes @ address 0x00000000 (Data = 0x20040000)
Read 4 bytes @ address 0x00000000 (Data = 0x20040000)
```

hubertmis, 2018-10-15:
@Nopik, Right, I've missed it. Not useful at all :)

The provided data does not indicate any obvious problem in the 802.15.4 driver.
I recommend to report this problem on Devzone (https://devzone.nordicsemi.com/). Our support should be able to help to investigate if the problem is in the application, the OS, the port or the driver. 
Note, that we have examples using the driver in FreeRTOS: https://infocenter.nordicsemi.com/index.jsp?topic=%2Fcom.nordic.infocenter.thread_zigbee.v2.0.0%2Fthread_client_server_example.html&cp=4_2_0_1_2_6 and we didn't hit reported problem during development.

Nopik, 2018-10-15:
Yeah, but you do realize, that if there is no radio activity, the problem does not appear? And, if the packets are sent at lower rate (much less than 50 packets/s), the problem shows up at longer time, too? I'm reporting it here, as I think that the driver does corrupt memory - so even if the hardfault is not inside the IRQ handler, the culprit might be there.

Do you have minimal example of using your driver directly with FreeRTOS? Or at least, without it?

The examples are using Thread, which is kind of different environment. My code basically does nothing particular:

```
  nrf_802154_init();
  nrf_802154_tx_power_set(0);
  nrf_802154_channel_set(26);
  uint8_t panId[] = { 0xb1, 0x04 };
  nrf_802154_pan_id_set( panId );
  nrf_802154_extended_address_set( newAddr );
  nrf_802154_receive();
```
and then in `nrf_802154_received`:
```
nrf_802154_buffer_free(p_data);
```

Is this approach correct? Maybe something is missing?

I dont do anything with the data (the code has some memcpy as a leftover, but commenting it out is not changing anything) - yet, if driver is not receiving data, hardfaults are not there.

Nopik, 2018-10-18:
So, I've been debugging it in meantime, and I think I've got near bottom of this problem. Indeed, the driver itself isn't too guilty here - though its default settings are one of the causes.

The story goes as this:

- there is `vApplicationIdleHook`, which does call `vTaskResume` on logger task. This hook is executing very very often (at most every few milliseconds, usually much more often).
- there is another task, radio listener, which only waits on mutex (it 'gets' the data and discard them). Not much going on here, but mutex itself is important here.
- Since there is a radio transmitter nearby, producing 20 packets/second, this mutex gets `GiveFromISR()` every 20 ms or so, from radio driver IRQ context (through `nrf_802154_received`).
- in FreeRTOS there is `pxReadyTasksLists` list of tasks, which are ready for execution. When task wakes up (either by `vTaskResume` or by mutex give), `prvAddTaskToReadyList` and then in turn `vListInsertEnd` gets called
- these functions are supposed to be called from critical section - and in fact, FreeRTOS code does have `taskENTER_CRITICAL();` in `vTaskResume` and its equivalent of `portSET_INTERRUPT_MASK_FROM_ISR();` in `xQueueGenericSendFromISR`
- the bug happens, because those critical section are not working properly - both paths can be executed simultaneously modifying the task ready list simultaneously, so the elements on the list gets garbled (which can be shown pretty well in gdb).
- once the list gets corrupted, all sorts of bad effects happen - straight from immediate hardfault (`pxCurrentTCB` becomes NULL, then task switching code is trying to dereference that and accesses invalid memory address), through memory corruption to weird task slowdowns.
- this only happens if tasks with equal priority try to become active at the same moment. If priorities are different, they are added to different lists, so no possibility of memory corruption

I will try to reproduce this problem with simple timer interrupt instead of radio driver - the code will be much shorter and problem easier to replicate.

Now, once I know what is happening, a number of workarounds is obvious. Like, use tasks with different priorities. Use event flags like nRF52840's port of OpenThread does - instead of mutex.

Another workaround is to modify radio driver settings - `NRF_802154_IRQ_PRIORITY` has default of `0`, and it allows for the bug to happen. Changing it to `1` does not help, but any value higher than `1` does make the problem go away.

I'm not sure if Nordic's FreeRTOS is buggy here (it shouldn't allow 2 simultaneous entrances to critical section? and e.g. nRF51 port seem to disable IRQ instead of raising priority), or just IRQ levels 0 & 1 should be used with much greater care.

Can you confirm, whether Nordic port of FreeRTOS is supposed to have IRQ 0 & 1 levels to be privileged enough to execute inside critical section of the OS? That basically means, they can't be used for any FreeRTOS calls, including `FromISR` API methods.

rakons, 2018-10-18:
Hi.
Inside FreeRTOSConfig that comes with examples you may find a definition:

```
/* !!!! configMAX_SYSCALL_INTERRUPT_PRIORITY must not be set to zero !!!!
See http://www.FreeRTOS.org/RTOS-Cortex-M3-M4.html. */
#define configMAX_SYSCALL_INTERRUPT_PRIORITY            configLIBRARY_MAX_SYSCALL_INTERRUPT_PRIORITY
```

It uses definition that is set little above in the same file:

```
/* The highest interrupt priority that can be used by any interrupt service
routine that makes calls to interrupt safe FreeRTOS API functions.  DO NOT CALL
INTERRUPT SAFE FREERTOS API FUNCTIONS FROM ANY INTERRUPT THAT HAS A HIGHER
PRIORITY THAN THIS! (higher priorities are lower numeric values. */
#define configLIBRARY_MAX_SYSCALL_INTERRUPT_PRIORITY _PRIO_APP_HIGH
```

Please consult app_util_platform.h to see the definiton of _PRIO_APP_HIGH, to make long story short, for CM4 it is set to 2, so this is exactly what you can observe.

Nopik, 2018-10-18:
Thanks, that was the last bit of information I was looking for.

closed 2018-10-18 by Nopik