NordicSemiconductor/nRF-IEEE-802.15.4-radio-driver #164: Max TX power is improperly handled

Nopik, 2018-10-19:
In the driver we have: https://github.com/NordicSemiconductor/nRF-IEEE-802.15.4-radio-driver/blob/master/src/nrf_802154_pib.c#L127 which allows user to set +9 dBm TX power. It seems, that this value, after normalization, gets directly to `NRF_RADIO->TX_POWER` register.

However, according to the nRF52840 spec (http://infocenter.nordicsemi.com/index.jsp?topic=%2Fcom.nordic.infocenter.nrf52840.ps%2Fradio.html&cp=2_0_0_5_19_13&anchor=register.TXPOWER) +8 is the max value. And, when `nrf_802154_tx_power_set` is called with a higher value, e.g. `10`, `+9dBm` is being set - but it results in very poor signal (we do measure like -65dBm or so). Setting +8 dBm seems to have correct result.

Can you verify this, please?

hubertmis, 2018-10-19:
Thanks for rising this issue.
You are right. The value +9 dBm is not valid for the Rev 1 version of nRF52840. I'll update the array to allow only valid values.

hubertmis, 2018-10-19:
@Nopik I've created the PR #165 . Could you have a look if it looks correctly to you?

Nopik, 2018-10-19:
Yes, LGTM.

PR #165 (2018-10-19): Update allowed TXPOWER values to match nRF52840 Rev 1 range
This PR fixes #164
closed 2018-10-19 by hubertmis