NordicSemiconductor/nRF-IEEE-802.15.4-radio-driver #166: nRF-IEEE-802.15.4 driver ported to mbedOS

markus-becker-tridonic-com, 2018-10-27:
Hi,

I am trying out whether it is possible to use this driver in combination with ARM mbedOS.

I am using the NRF52840_DK target and disabled SOFTDEVICE. After adding a few files/directories to an .mbedignore file, I got the mbed-os-example-client to compile and link. Then I wrote a small adaptation layer for the 6LoWPAN stack of mbedOS. It seems to work so far, that the 6LoWPAN stack wants to send a radio packet. The packet never makes it to radio though.

The problem seems to be an init/timing problem:
* HFCLK is started only by all_prec_request() through nrf_802154_core_transmit() within critical section
* in tx_init() timeslot_is_granted() returns false because m_rsch_timeslot_is_granted is false
* tx_init returns false and does not send
* POWER_CLOCK_IRQHandler() calls nrf_802154_clock_hfclk_ready()
* nrf_802154_critical_section_exit()
* m_rsch_timeslot_is_granted is set to true

Am I missing something a pre-requisite for transmission?

Attached find the message sequence chart:
[image]

hubertmis, 2018-11-05:
Thanks for reporting this problem. It looks that CSMA-CA procedure skips first transmission attempt if CSMA-CA is requested in the sleep state (because radio access is not granted yet by the arbiter). We should fix this bug, but it does not looks as a root cause of your problem.

The diagram shows that the program hangs after first transmission attempt and there are no more transmission attempts. Could you provide a backtrace from GDB captured when the problem occurs? It is hard to tell what is going wrong with only the sequence diagram.

markus-becker-tridonic-com, 2018-11-08:
I have moved from `nrf_802154_transmit_csma_ca()` to `nrf_802154_transmit()` (with cca=false for the moment) and then I do not have this problem anymore. The packet is actually transmitted then. Did not investigate what the difference between the two methods is.

hubertmis, 2018-11-13:
The `nrf_802154_transmit_csma_ca()` function performs CSMA-CA procedure (automatic transmission retires when if the channel is busy). The `nrf_802154_transmit()` performs single transmission.

I'm closing this issue as you don't have the problem anymore. If you hit this problem again, please reopen.

Out of curiosity, are you going to integrate your work to the official mbedOS repository? Or is it just a private project?

closed 2018-11-13 by hubertmis