NordicSemiconductor/nRF-IEEE-802.15.4-radio-driver #167: Application blocks in nrf_802154_init() functions

Zhiyi-Zhang, 2018-11-01:
I tried the examples given on the wiki page, however, I met some issues in nrf_802154_init() functions.
I found my program stops at this line:
```
nrf_802154_clock_init();
```

If I comment this line out, then the program stops at this line:
```
 nrf_802154_lp_timer_init();
```

**The environment:**
I am using the source file nrf_802154_clock_nordrv.c,
Compiling with -DNRF52840_XXAA
and -DRAAL_SINGLE_PHY

**Hardware:**
nrf52840 DK

Any ideas to help?

hubertmis, 2018-11-05:
Thanks for reporting this problem.
`nrf_802154_lp_timer_init()` function hangs if the clock driver is not initialized, because it waits for the low-frequency clock here: https://github.com/NordicSemiconductor/nRF-IEEE-802.15.4-radio-driver/blob/master/src/platform/lp_timer/nrf_802154_lp_timer_nodrv.c#L399 .

I can't find any potential software issues in `nrf_802154_clock_init()` function (like endless loops or waiting for events). Have you tried other board to rule out hardware issue?

Zhiyi-Zhang, 2018-11-07:
> Thanks for reporting this problem.
> `nrf_802154_lp_timer_init()` function hangs if the clock driver is not initialized, because it waits for the low-frequency clock here: [...]
No. I haven't tried other boards. However, I successfully run the test with the template project on https://github.com/dpiegdon/nordic-template.
Does this mean that the driver must work with CMSIS or nrfx to function?

ochipara, 2018-11-26:
I am running into the same problem of blocking in the nrf_802154_clock_init(). Is there a simple "hello world" sample that sends and receives with the appropriate dependencies?

hubertmis, 2018-11-27:
@ochipara Currently we do not provide any build system with dependencies to create a sample application based on the 802.15.4 driver. 
Have a look at this project: https://github.com/dpiegdon/nordic-template/tree/example-802.15.4 . There is a simple build system including the driver and its dependencies.

Could you provide some details how to reproduce your problem? How can we build the application that blocks in the `nrf_802154_clock_init()` function?

ochipara, 2018-11-27:
Dear hubertmis, 

I think that the problem that I was having was that I was not configuring the board's clock correctly. It was not immediately clear (sorry, new to NRF's boards) how to appropriately configure the various hardware settings. I got it working in the end. I have my example at https://github.uiowa.edu/ochipara/nrf52-802154. It uses the driver + NRF's SDK. I assume it is fairly similar to your example.

A couple of remarks
- The sample on the wiki for the receiver and sender is out of date. It uses an older API where the length was included as one of the arguments.
- It might be nice to provide a struct that separates out the length of the packet from its content rather than simply representing the packet as a uint8_t *.

hubertmis, 2018-11-28:
That's good to know that you managed to run the driver. Unfortunately, I don't have access to the project you linked above.

We provide two APIs for frame formats:
* RAW functions (enabled by default) take as an argument a buffer directly provided to the EasyDMA controller.
* 'normal' functions (used by the examples from wiki) take two arguments: PSDU buffer and PSDU buffer length.
I'll add explanation to wiki that provided examples use 'normal' API and it should be selected in the configuration.

I like the idea of the packed struct instead of `uint8_t *`. We'll evaluate this API modification for version 2.0 of the driver.

It seems that the issue is solved. I'm closing it. Feel free to reopen if anything else should be provided to prevent this problem.

closed 2018-11-28 by hubertmis