NordicSemiconductor/nRF-IEEE-802.15.4-radio-driver #177: Check destiny PAN Id function not working

jmgaraujo, 2018-11-30:
I'm testing the driver to integrate with one of our products and the function static `bool dst_pan_id_check(const uint8_t * p_psdu)` is returning false where the input data is correct. 
The reason is that  `PAN_ID_OFFSET` is set to 4 and then when reading the data `&p_psdu[PAN_ID_OFFSET]` it reading from the wrong position. 

The comment says `///< Offset of Pan Id in Data frame (+1 for frame length byte).` but then when accessing the buffer it should decremented `PAN_ID_OFFSET` by 1. 

I've defining `PAN_ID_OFFSET` to 3 and now I'm receiving packets, but I'm not doing a pull request because  I'm not sure if this is going to impact somewhere else.

hubertmis, 2018-11-30:
@jmgaraujo , thanks for reporting this problem. 
Are you using this driver to transmit frames or some third-party transmitter? Could you share an example of transmitted and received data? Are you using `nrf_802154_received()` or `nrf_802154_received_raw()` function?

jmgaraujo, 2018-11-30:
To transmit frames I'm using a third party transmitter and validating the frames using the SmartRF Packet Sniffer 2 with Wireshark. 

Received data on the function `dst_pan_id_check`:
`p_psdu = 13 00 A9 46 6A FF FF 46 6A 46 6A 00 06 5E A4 03 E7 01 3C 72`

Transmitted data, captured on the Wireshark: 
`Wireshark = 00 a9 46 6a ff ff 46 6a 46 6a 00 06 43 55 03 e7 01 b0 30`
[image]

I'm using `nrf_802154_received()`.

hubertmis, 2018-11-30:
@jmgaraujo Thanks for the details. The problem is that you are using 2015 version of 802.15.4 frames. The driver right now fully support version 2006. There is the 'sequence number suppression' bit set in the transmitted frame. This 2015 feature is not implemented in the driver yet.
Modification of the PAN ID offset may work correctly for now, but if you receive a frame without 'sequence number suppression', the offset will be invalid again.

jmgaraujo, 2018-11-30:
That makes total sense. 
But I thought that the driver was supporting the 2015 version by looking at the function `bool frame_type_and_version_filter(uint8_t frame_type, uint8_t frame_version)` where it is accepting all versions up to `FRAME_VERSION_2`. 

Do you have any idea when you are going to have support for 2015 version?

hubertmis, 2018-12-03:
Right now there is only basic support of 2015 feature in the driver. There is a PR #146 that includes support for sequence number suppression. I'm going to work to merge this PR soon.

closed 2019-01-03 by hubertmis

hubertmis, 2019-01-03:
@jmgaraujo , according to our internal tests the sequence number suppression bit should be correctly parsed by the 802.15.4 driver now. Could you try to rerun your tests with current master branch?

jmgaraujo, 2019-01-03:
Excellent, it's working now. Thank you!

PR #179 (2019-01-03): Enhanced Acks generator
This PR provides enhanced Ack support, better performance of the Precise Ack Timeout module and improved support of other features present in the 802.15.4-2015 specification.

This PR should fix #177.
I based this patch on #146.