NordicSemiconductor/nRF-IEEE-802.15.4-radio-driver #201: Integer overflow in nrf_802154_swi.c

e-rk, 2019-02-21:
The definitions of functions in `nrf_802154_swi.c`:
* `queue_ptr_increment`
* `queue_is_full`
take `uint8_t queue_size` argument. This causes integer overflow when `NRF_802154_RX_BUFFERS` is larger than `252`.

hubertmis, 2019-02-21:
@e-rk thanks for the bug report. For the clarification: do you need larger buffer than 252 with `nrf_802154_swi.c` enabled?

e-rk, 2019-02-22:
Thanks for mentioning that `nrf_802154_swi.c` can be disabled. I used the `direct` variant instead.

I had to add some missing source files for that configuration:
```
nrf_802154_frame_parser.c
nrf_802154_ack_data.c
nrf_802154_ack_generator.c,
nrf_802154_enh_ack_generator.c
nrf_802154_imm_ack_generator.c
```
Then it compiled fine.