NordicSemiconductor/nRF-IEEE-802.15.4-radio-driver #217: Disable NRF_802154_TIMER_INSTANCE for Rx only devices

jmgaraujo, 2019-03-29:
Hi, 
After a quick read through the code I  saw the time NRF_802154_TIMER_INSTANCE is only used for the FEM and ACK IFS which are features that I'm not using. 
For this project the 802.15.4 driver is only used to receive Beacon packets and will never be required to transmit ACK or Data packets. We also don't use the FEM feature. 

That said do you think it's possible to spare the timer NRF_802154_TIMER_INSTANCE? Or are there other features that depend on it?  

If is possible to spare this driver, are there any PPI lines that will also be free by consequence? 

Best Regards, 
José

hubertmis, 2019-03-29:
@jmgaraujo , you are right. `NRF_802154_TIMER_INSTANCE` is used only for the FEM and Ack frames.

It looks that in your application you can safely remove code related to this timer and all PPIs except `PPI_DISABLED_EGU` and `PPI_EGU_RAMP_UP` - these two PPIs are used to force transitions of RADIO FSM.
Also, there are some PPIs used to fetch timestamps of received frames. I'm not sure if you need these.

jmgaraujo, 2019-03-29:
@hubertmis thank you for your quick reply. 
Yes, I'm using the timestamps of the received frames.

Is this something that can be added as a feature request for the driver?

hubertmis, 2019-03-29:
@jmgaraujo, I'll talk to PO and get back to you. For sure we don't plan such feature any time soon.

hubertmis, 2019-04-11:
@jmgaraujo, we've discussed this request and decided that RX-only configuration that releases some hardware peripherals is an interesting feature. However, due to limited resources in our team, it's not something we want to officially support. Because of that, we don't plan to add such feature to the driver.

jmgaraujo, 2019-04-11:
@hubertmis, thank you for the feedback.

hubertmis, 2019-05-10:
@jmgaraujo, do you need any other assistance regarding this topic, or shall we close this issue?

jmgaraujo, 2019-05-10:
@hubertmis, no thank you. You can close it.

closed 2019-05-10 by hubertmis