NordicSemiconductor/nRF-IEEE-802.15.4-radio-driver #218: import driver to new project, how to include sdk?

rs-development, 2019-04-01:
What is the best way to start a new project with this driver?

I am kinda lost with the documentation for the nRF52840 DK. The examples provided with the SDK are overwhelming and i am not getting into it at all. 

Is there a description how to include this driver and the SDK to a new project?
Since it's recommended i am using SEGGER Embedded Studio right now.

hubertmis, 2019-04-02:
I suggest to have a look at this project: https://github.com/dpiegdon/nordic-template/tree/example-802.15.4 . It provides simple example how to build an application based on this driver.

If you want to integrate the driver with SDK, I recommend to ask https://devzone.nordicsemi.com . Our support team will help.

rs-development, 2019-04-04:
I have taken a look into the project and it kinda helped but it also left me a bit confused. I have taken a look how the driver is integraded into openthread aswell.

For example the  nrfx/hal/nrf_radio.h file exists also in the driver at /src/hal/nrf_radio.h. What is the intention behind this? And does the nrfx still need the radio.h file provided by the nrfx or is it fine to build only the radio.h from the driver?

hubertmis, 2019-04-04:
When we created this driver, there was no `nrf_radio.h` file in nrfx and we created our own. We plan to use the `nrf_radio.h` file from nrfx since it was created, but right now it is unused. You should use `nrf_radio.h` from `src/hal/`, the other one can be safely removed from your project.

hubertmis, 2019-04-15:
FYI, with #221 we removed internal `nrf_radio.h` and now we use only hal headers from nrfx. 

Is your problem solved, or do you need any more assistance?

rs-development, 2019-04-17:
Thanks for proting the dirver to the SDK 👍 
I got the project together and started testing this morning. but i already reconised some problems. See the other issue. #223

hubertmis, 2019-05-10:
@rs-development , can we consider this problem solved now?

rs-development, 2019-05-13:
@hubertmis yes

closed 2019-05-13 by rs-development