NordicSemiconductor/nRF-IEEE-802.15.4-radio-driver #219: how can I debug my project in keil?

gksrb4, 2019-04-04:
my application is hanging sometimes. (this means no response. It is necessary to analyze the phenomenon more accurately and deeply.) To solve this issue, I want to change optimization level 0. for now it set 3.

I set 0. compile adding preprocessor ENABLE_DEBUG_ASSERT=1 DEBUG and program to my board. but it is always hanging timeslot routine.
[image]

hubertmis, 2019-04-04:
@gksrb4 it is known problem that this loop https://github.com/NordicSemiconductor/nRF-IEEE-802.15.4-radio-driver/blob/186919244df3f58122111a91d2470d3c2177a109/src/platform/lp_timer/nrf_802154_lp_timer_nodrv.c#L520 is indefinite without compiler optimizations. I'm not familiar with Keil optimization levels. Most compilers allows to enable some optimizations 'debug optimization level' to run code much faster while making debugging easy. 

However, if it is not possible to enable such level in Keil, I suggest to change the loop condition to `while (0)` during debugging sessions. With this patch you don't get proper timestamps of received frame, but the driver would not stuck in indefinite loop.

gksrb4, 2019-04-10:
hmm.. okay , Is this going to be fixed in the future?

hubertmis, 2019-04-10:
@gksrb4, it is hard to tell. From the top of my head, I don't see any easy fix. We could rewrite this loop in assembler, but it would be not portable and hard to maintain. Moreover, this code is in the `platform` directory that contains example of low-power timer driver that provides API for the 802.15.4 driver - it is not in the 802.15.4 driver code that we focus on mostly.

Anyway, I'll discuss it with PO and keep you posted.

gksrb4, 2019-04-11:
Thank you for your kind answer.

closed 2019-04-11 by gksrb4

hubertmis, 2019-04-11:
We decided to add support for build without optimizations to the product backlog. I cannot commit to any date right now, but it will be fixed in the future.