NordicSemiconductor/nRF-IEEE-802.15.4-radio-driver #226: frame length incorrect if PANID compression is not used

rs-development, 2019-04-30:
While receiving frames with NOT activated PANID compression i realized the indicated frame length in the first field of pdata is incorrect.

I have compared the data received from the nrf52840 with the data received from an SDR. The SDR is able to decode the frame correctly. The driver on the nrf52840 is not. It is not receiving the full frame length, is seens to be ignoring the that PANID compression is not enabled.

Any suggestions how to make this work?

hubertmis, 2019-04-30:
@rs-development, thanks for reporting this issue. Could you provide an example of a frame that is incorrectly decoded?

rs-development, 2019-04-30:
You got me, while preparing the screenshots i recognized some mistake in the mac layer frame length calculation, I messed up between hex and dec values. It seems to be working fine. Frame length is increased by 2 octets. Great work on your side, not so good on mine.

closed 2019-04-30 by rs-development