NordicSemiconductor/nRF-IEEE-802.15.4-radio-driver #227: Continuous carrier power setting

rs-development, 2019-04-30:
Me again...

I am trying to evaluate different power settings in the continous carrier mode. I am using a power meter connected to the SWF/SDW connector.

Since i think the power settings in from the pib are only applied while changing state, i added a state change to sleep and back to receive before the continuous carrier is started.

But weather i use a output power of -20, 0 4 or 8dBm i can only ever measure -2dBm at the output.

    nrf_802154_init();
    nrf_802154_channel_set(CHANNEL);
    nrf_802154_tx_power_set(-20);
    // state change so the pib values are applied
    nrf_802154_sleep();
    nrf_802154_receive();
    nrf_802154_continuous_carrier();

Any ideas on this one?

rs-development, 2019-04-30:
While sending normal frames the power setting seems to be working good. If i am sending frames every 25ms and measuring the average output power, i am getting plausible results.

-37dBm with 0dBm power setting
-55dBm with -20dBm power setting

So how can i apply the power setting to the continuous carrier?

hubertmis, 2019-04-30:
@rs-development, I don't think we have ever considered changing power of the continuous carrier transmission. It seems that this feature is not implemented. 

Have a look at the `continuous_carrier_init()` function in the `nrf_802154_core.c`. It looks that `nrf_radio_txpower_set(nrf_802154_pib_tx_power_get());` instruction is missing. You could try to add it like in the `tx_init()` function.

Are you willing to create a PR that adds this feature?

rs-development, 2019-04-30:
If i get this going i will add a PR

closed 2019-05-10 by hubertmis