NordicSemiconductor/nRF-IEEE-802.15.4-radio-driver #230: Get RX Event

warner83, 2019-05-05:
Is there a way to catch the beginning of the RX operations? Through a callback or simply by polling radio status.
I tried by reading the NRF_RADIO->STATE but the state does not change during packet reception.

hubertmis, 2019-05-05:
@warner83, in current implementation the driver does not handle an event indicating start of a received frame and because of that you cannot instantly get a notification. 

Why do you need to catch this event? I hope most requirements can be met with current driver design and API.

warner83, 2019-05-07:
I'm currently integrating the driver into Contiki-NG. The OS implements CSMA/CA and TSCH, however, such code requires to catch the beginning of the reception of a packet from the radio. I modified the driver to capture the FRAMESTART event and seems to work.

warner83, 2019-05-09:
@hubertmis, as you were mentioning the FRAMESTART event is not delivered instantly but after some time. Do you have any suggestion to catch the event?

hubertmis, 2019-05-10:
@warner83 , `FRAMESTART` should be triggered right after PHR is transmitted or received.

Why is it required by Contiki to catch event at the beginning of the frame? Is it used to get a timestamp of the frame (needed by TSCH)? We provide another solution to get precise timestamps of received frames.

warner83, 2019-05-10:
It looked like the event was triggered with some delay, I'll check it out again. 

The TSCH implementation in contiki has some kind of active wait for packet reception, so I need to catch the beginning of the packet reception. 
I saw the callback nrf_802154_received_timestamp, where a timestamp of the end of reception is provided in us

hubertmis, 2019-05-12:
@warner83 , I don't know Contiki implementation, but I can imagine that Contiki MAC does not disable receiver if it detects that a frame is being received at the end or reception window. Receiver may be disabled after this frame is received.

We are implementing similar functionality called delayed rx. We have in mind CSL and TSCH while we design this feature. I think you could have a look if delayed rx can be used instead of typical receive procedure for TSCH. Delayed rx should keep much more strict timings than typical rx managed by MAC layer. Note that this feature is still in experimental state.

hubertmis, 2019-06-10:
@warner83, is your problem solved now? Or do you need any more assistance?

warner83, 2019-08-01:
Unfortunately I went on implementing other functionalities. I'll go back now to my original issue. I'll let you know. Thanks!

hubertmis, 2019-08-14:
It seems all questions were answered so far and there was no discussion for some time now. I'm closing this issue. @warner83 , if you have any more questions feel free to reopen.

closed 2019-08-14 by hubertmis