NordicSemiconductor/nRF-IEEE-802.15.4-radio-driver #235: porting for the nrf52811

rs-development, 2019-05-22:
Is there any work in progress for porting the driver to the nRF52811?

We are currently working on the nRF52840 DK but would like to move to our own hardware soon which is using the nRF52811.

hubertmis, 2019-05-22:
It's ported since version 1.3.0: https://github.com/NordicSemiconductor/nRF-IEEE-802.15.4-radio-driver/blob/master/release_notes/nrf_802154_1.3.0.md#new-features

hubertmis, 2019-05-22:
@rs-development, have you found misleading info somewhere?

rs-development, 2019-05-22:
Ahh i would not call it that way :smiley:

But my last information was the SDK is not ported yet and i assumed the same of this driver. The repositroies README file did not say something else, maybe just add it for the slower ones like me .

> The nRF IEEE 802.15.4 radio driver implements the IEEE 802.15.4 PHY layer on the Nordic Semiconductor nRF52840 SoC.

hubertmis, 2019-05-22:
Oh, right, I'll update it. Probably some more invalid info can be found in Wiki, I'll have a look searching for "52840" :)

rs-development, 2019-05-22:
👍 

I leave it open open for you to close it.

PR #237 (2019-05-24): Add nRF52811 to supported SoCs list in README.md
Fixes #235
closed 2019-05-24 by hubertmis