NordicSemiconductor/nRF-IEEE-802.15.4-radio-driver #239: Question - FEM (front end module) control with single pin for LNA and PA

rs-development, 2019-05-23:
I would like to use a FEM with the nRF.

I checked and the driver and it seems to have working fem control implemented. It uses separate pins for LNA and PA.

Is there a disadvantage if only one pin is used to toggle between PA and LNA, except from the the LNA being always on if the radio is not transmitting?

hubertmis, 2019-05-23:
I think you should verify in the datasheet of the FEM you are using. From the perspective of the 802.15.4 driver there are no disadvantages. But I imagine that analog front-end may have some limitations.

hubertmis, 2019-08-13:
@rs-development , can we consider this issue solved, or do you need any more assistance?

hubertmis, 2019-08-14:
The question is answered and the thread was inactive for a few months. I'm closing the issue. @rs-development , please reopen if you have any more questions.

closed 2019-08-14 by hubertmis