NordicSemiconductor/nRF-IEEE-802.15.4-radio-driver #248: Question about known issue "Deinitialization of the driver..."

jmgaraujo, 2019-06-12:
Hi,
Can you please provide more information about the known issue "Deinitialization of the driver works incorrectly in multiprotocol configuration (simulator or SoftDevice RAAL)."?

We are using the multiprotocol configuration with the SoftDevice and there is a new feature that we are working on that will need to support the device to operate with Bluetooth only or with in the multiprotocol mode with Bluetooth and IEEE 802.15.4. 

Is there a workaround for this issue and a timescale for the fix? 

Regards,
José

hubertmis, 2019-06-12:
@jmgaraujo In typical applications you don't need to deinitialize the 802.15.4 driver. You just put 802.15.4 to sleep when you use Bluetooth only. When you need to use 802.15.4 you put the 802.15.4 driver to the rx or the tx state.
Is there any reason why you would like to uninitialize the driver?

jmgaraujo, 2019-06-12:
@hubertmis I was assuming that uninitialize the driver was the best way to disable it, but sounds that putting the driver to sleep is enough for our application. 
The main reason to do this is to save power and leave the radio 100% of the time allocated to the Bluetooth.
Thank you for your time.

closed 2019-06-12 by jmgaraujo