NordicSemiconductor/nRF-IEEE-802.15.4-radio-driver #249: Newlib rand() is not interrupt-safe

avtolstoy, 2019-06-13:
### Problem

`rand()` should not be used from an ISR because it relies on a thread-local reentrant [struct](https://github.com/eblot/newlib/blob/master/newlib/libc/include/sys/reent.h#L314): https://github.com/NordicSemiconductor/nRF-IEEE-802.15.4-radio-driver/blob/master/src/mac_features/nrf_802154_csma_ca.c#L141 https://github.com/eblot/newlib/blob/master/newlib/libc/stdlib/rand.c#L89

While there is an issue of interrupt-safety when accessing/modifying `_rand48`, the main problem we are seeing is that depending on the current active thread at the time the interrupt arrives, `newlib` may attempt to allocate this structure from heap, which is not allowed at least on our platform and in general is a very bad idea.

I see that now it's quite easy to implement our own `nrf_802154_random_get()` to use for example `rand_r`, however I think the default implementation should also behave safely.

hubertmis, 2019-06-13:
Thanks for reporting this issue. C standard library provides only `rand()` function as a random number generator. However, as you noticed, this function is not obligated to be reentrant / thread-safe. In the default implementation we have provided we wanted to use only standard library functions.
I agree that the default implementation should be safe. That's why I think we could create a new implementation of random platform that would use newlib's (probably most popular embedded c std lib) `rand_r` instead.
What do you think about this solution?

avtolstoy, 2019-06-14:
:+1: Sounds good to me. Thanks for looking into this!

PR #268 (2019-08-27): Re-entrant random number generator
Fixes #249
closed 2019-08-27 by hubertmis