NordicSemiconductor/nRF-IEEE-802.15.4-radio-driver #351: Stops receiving data after 120s when SD send event NRF_EVT_RADIO_SESSION_IDLE

cjsueos, 2020-03-24:
Hi,
Setup: nrf52840 running SD s140 version 7.0.1
nRF-IEEE-802.15.4-radio-driver master branch.
FreeRTOS from NRF SDK 16
BLE is not enabled for now.

Our application is default listening for 802.15.4 data and it works the first 120secs from startup. After about 120secs SD sends a NRF_EVT_RADIO_SESSION_IDLE and after that we won't get any more data. Only thing is to call _sleep and then _receive again to start listening. Just calling _receive doesn't help.
It does the same after FDS has written a page to flash.

Shouldn't the nrf 802.15.4 driver keep requesting timeslots while in _receive mode?
Or is there anything that I've missed in the documentation?

hubertmis, 2020-03-24:
@cjsueos , it sounds like the 802.15.4 driver is not notified that it can request a new timeslot after the previous one is finished. Please, make sure that this part is implemented in your application:

> This arbiter depends on SoC events from the SoftDevice. They have to be supplied to the driver using the nrf_raal_softdevice_soc_evt_handler function. For more information about reading SoC events, refer to the SVC call sd_evt_get.

cjsueos, 2020-03-25:
@hubertmis Thank you! Works perfectly after adding a soc event observer.

closed 2020-03-25 by cjsueos