NordicSemiconductor/nRF-IEEE-802.15.4-radio-driver #358: Wiki is gone?

jalgroy, 2020-06-23:
Hi!
If I'm not mistaken there was a Github wiki in this repo with examples of transmitting/receiving using this driver, but now it seems to be gone. Are the examples available anywhere else?

Thanks.

hubertmis, 2020-06-23:
@jalgroy , Github wiki was outdated. Because of that we decided to disable it. At the moment we are preparing up-to-date documentation.

> Are the examples available anywhere else?

Not at the moment. The examples were outdated and not working correctly. I recommend to use API documentation from https://github.com/NordicSemiconductor/nRF-IEEE-802.15.4-radio-driver/blob/master/src/nrf_802154.h until high-level documentation with examples is ready.

hubertmis, 2020-07-06:
The documentation is back online. Please check this link: https://infocenter.nordicsemi.com/index.jsp?topic=%2Fstruct_drivers%2Fstruct%2Fradio_driver_latest.html

closed 2020-07-06 by jalgroy