NordicSemiconductor/nRF-IEEE-802.15.4-radio-driver #359: where does nrf_error.h come from?

rs-development, 2020-07-05:
Where does nrf_error come from? I use the library in conjunction with the nrfx driver repository, which does provide the nrf_radio.h but I can not find a nrf_error.h header file. Only nrfx_error.h

hubertmis, 2020-07-06:
@rs-development , this issue was already discussed in #354 . If I remember correctly this error code comes from SoftDevice headers.

rs-development, 2020-07-09:
That seems fine. 

But I ran into some more trouble. For example:

In src/mac_features/nrf_802154_ack_timeout.c the nrf_802154_timer_sched_remove function call in line 118 only provides one argument. But the nrf_802154_timer_sched_remove function from nrf_802154_timer_sched.c need 2 arguments.

The same problem occurs at some other places for example at the core file:
```
static void rssi_measure(void)
{
    m_flags.rssi_started = true;
    nrf_radio_event_clear(NRF_RADIO_EVENT_RSSIEND);
    nrf_radio_task_trigger(NRF_RADIO_TASK_RSSISTART);
}
```

But here I assume I am using the wrong version of the nrfx drivers. 

Can you tell which version of nrfx should be used? 
And what is wrong with the first problem?

Thanks mate!